sanny32/OpenModSim #3: Selection of network card

Comiter, 2024-03-14:
I'm trying to configure the Modbus slave to be active on a specific NIC om my computer, but I'm not able to select which NIC the slave should connect to, I'm only able to select which port it should listen to.
Is there a setting in any of the configuration files, or is the IP selected  based on first NIC with connection?

sanny32, 2024-03-14:
Hello @Comiter. I`m sorry, you can`t configure specific NIC in Open ModSim utility.

Comiter, 2024-03-15:
> Hello @Comiter. I'm sorry, you can't configure specific NIC in Open ModSim utility.

Ah, okay. So the OpenModSim is not able to reply to requests from network traffic at all? It is only designed to test applications on local computer?

sanny32, 2024-03-15:
No, Open ModSim accept connection from any network interface.