sanny32/OpenModSim #7: Modbus-RTU-4ch-Relay-12v

cofhal, 2024-06-03:
I have been trying to use this program with this card:
https://mikroelectron.com/Product/Modbus-RTU-4ch-Relay-12v-RS485-Module-4-Input-Optocoupler-Isolation/
 but it does not work. What could I be doing wrong or if it definitely doesn't work for this type of cards.

sanny32, 2024-06-03:
Hello @cofhal. For answer to you need more information. 
1. Connection scheme and modbus params. 
2. Operation system where execute OpenModSim. 
3. Serial port log received from any serial port sniffer program.

talenat-st, 2024-11-03:
@cofhal you can not use modbus slave simulator for that kind of card.You need modbus master so look at this programm https://github.com/sanny32/OpenModScan

closed 2025-06-21 by sanny32