sanny32/OpenModSim #9: missing Address 42001 value 0x0008

Chadlucc, 2024-07-24:
[image]
[image]

sanny32, 2024-07-24:
This happens because the read request comes for the address starting with 4x2002. The address numbering in the program OpenModSim starts with 1, not 0.
At the same time, in the protocol logs, addressing starts from 0, as provided in the Modbus protocol standard.

sanny32, 2024-10-24:
From version 1.6.0 OpenModsim support address base switch. Now you can select which address base use 0 or 1.