sanny32/OpenModSim #14: Problem with compilation

UrbanBobek, 2024-07-26:
Hi,

I would like to try out the simulator, but I am having difficulties compiling the program. 

I am on a Linux machine (PopOS). I have installed qt5
```
$qmake -version
QMake version 3.1
Using Qt version 5.15.3 in /usr/lib/x86_64-linux-gnu
```

When I try compiling the program I get the following error:
```
qmake omodsim.pro 
Project ERROR: Unknown module(s) in QT: help
```
From what I have read, the help module should be available with QT 5.13 

If it helps, this is the output of `qmake -query`
```
 qmake -query                                                                                                                                                                                
QT_SYSROOT:                                                                                                                                                                                   
QT_INSTALL_PREFIX:/usr                                                                                                                                                                        
QT_INSTALL_ARCHDATA:/usr/lib/x86_64-linux-gnu/qt5                                                                                                                                             
QT_INSTALL_DATA:/usr/share/qt5                                                                                                                                                                
QT_INSTALL_DOCS:/usr/share/qt5/doc                                                                                                                                                            
QT_INSTALL_HEADERS:/usr/include/x86_64-linux-gnu/qt5                                                                                                                                          
QT_INSTALL_LIBS:/usr/lib/x86_64-linux-gnu                                                                                                                                                     
QT_INSTALL_LIBEXECS:/usr/lib/x86_64-linux-gnu/qt5/libexec                                                                                                                                     
QT_INSTALL_BINS:/usr/lib/qt5/bin                                                                                                                                                              
QT_INSTALL_TESTS:/usr/tests                                                                                                                                                                   
QT_INSTALL_PLUGINS:/usr/lib/x86_64-linux-gnu/qt5/plugins                                                                                                                                      
QT_INSTALL_IMPORTS:/usr/lib/x86_64-linux-gnu/qt5/imports                                                                                                                                      
QT_INSTALL_QML:/usr/lib/x86_64-linux-gnu/qt5/qml                                                                                                                                              
QT_INSTALL_TRANSLATIONS:/usr/share/qt5/translations
QT_INSTALL_CONFIGURATION:/etc/xdg
QT_INSTALL_EXAMPLES:/usr/lib/x86_64-linux-gnu/qt5/examples
QT_INSTALL_DEMOS:/usr/lib/x86_64-linux-gnu/qt5/examples
QT_HOST_PREFIX:/usr
QT_HOST_DATA:/usr/lib/x86_64-linux-gnu/qt5
QT_HOST_BINS:/usr/lib/qt5/bin
QT_HOST_LIBS:/usr/lib/x86_64-linux-gnu
QMAKE_SPEC:linux-g++
QMAKE_XSPEC:linux-g++
QMAKE_VERSION:3.1
QT_VERSION:5.15.3
```
Thanks,
Urban

sanny32, 2024-07-26:
Hello @UrbanBobek. I suppose you must install `qttools5-dev` package.

UrbanBobek, 2024-07-26:
Hi @sanny32 , thank you for the quick response!
Yes, I was missing this package... I must have missed it when installing QT. Thank you!

A side note, after running make I get this error
```
mv -f /home/ubobek/ws/rpi/test/OpenModSim/omodsim/docs/jshelp.qhc /home/ubobek/ws/rpi/test/OpenModSim/omodsim/docs && mv -f /home/ubobek/ws/rpi/test/OpenModSim/omodsim/docs/jshelp.qch /home/ubobek/ws/rpi/test/OpenModSim/omodsim/docs
mv: '/home/ubobek/ws/rpi/test/OpenModSim/omodsim/docs/jshelp.qhc' and '/home/ubobek/ws/rpi/test/OpenModSim/omodsim/docs/jshelp.qhc' are the same file
make: *** [Makefile:525: omodsim] Error 1

```

I assumes the documentation fails to build? The program works regardless.

sanny32, 2024-07-29:
I assume that this is related to the folder where the project is built. The following code fragment of `omodsim.pro` file is responsible for this
```
# Copy doc files
DOC_FILES = $$PWD/docs/jshelp.qhc \
            $$PWD/docs/jshelp.qch
for(file, DOC_FILES) {
    win32:file ~= s,/,\\,g
    !isEmpty(QMAKE_POST_LINK) QMAKE_POST_LINK += &&
    QMAKE_POST_LINK += $$QMAKE_MOVE $$quote($$file) $$quote($$OUT_DOCS)
}
```

UrbanBobek, 2024-07-29:
I see. Thank you for your support!

closed 2024-07-29 by UrbanBobek