sanny32/OpenModSim #17: Set Auto simulation to random for all registers

parthitce, 2024-08-12:
Thanks for the great effort, we are planning to move from ModSim windows to this open source in Linux.

Question:
Is there a way to control/configure multiple registers at once? For example, set all or group of registers to auto simulation random?

parthitce, 2024-08-12:
> Hello @parthitce. There is no multiple registers simaltion feature. I think it can be added in future release.

Many thanks for quick response.

sanny32, 2024-08-13:
As workaround yoy can write script like this
```javascript
for(i = 1; i <=100; i++)
{
    Server.writeHolding(i, Math.random() * 65535);
}
```
and run it periodically.

parthitce, 2024-08-13:
> As workaround yoy can write script like this
> 
> ```js
> for(i = 1; i <=100; i++)
> {
>     Server.writeHolding(i, Math.random() * 65535);
> }
> ```
> 
> and run it periodically.

Thanks, haven't explored the scripts part yet. Will do!