sanny32/OpenModSim #18: Unable to set different device id for different modsim instances.

evo0705, 2024-09-23:
Somehow, when changing the device id in ModSim2, it changes the device id in ModSim1 to the same Id.
[image]

sanny32, 2024-09-23:
Hello @evo0705. This is the “feature”. Because Qt Modbus stack doesn’t allow use multiple id's.

talenat-st, 2024-11-03:
@sanny32 this is really shame as your programm is superb and multiple slaves (addresses) would be really really needed.
I use OpenModSim for testing and find it comparable to ModBus slave except this "feature" that limits use.
Please add this to new release

sanny32, 2025-06-21:
This will require rewriting the modbus stack from the qt library, which I would not like to do.

sanny32, 2025-06-22:
Of course, there is no need to rewrite the entire stack. But the functionality responsible for processing incoming modbus requests is in the private Qt class and cannot be redefined from the outside (see https://github.com/qt/qtserialbus/blob/dev/src/serialbus/qmodbustcpserver_p.h#L72). Therefore, there is a need to rewrite the stack in terms of processing all incoming requests. And this entails moving the entire stack of qt inside the program.

raspopov, 2025-09-18:
> But the functionality responsible for processing incoming modbus requests is in the private Qt class and cannot be redefined from the outside 

This is not entirely true. We can replace ModbusTcpServer's internal object, `d_ptr`, with our own one by using its `swap()` method and re-assembling it with private classes from the Qt source code changing for example the `QModbusTcpServerPrivate::matchingServerAddress()` method.

closed 2025-10-07 by sanny32

raspopov, 2025-10-16:
I would like to express my sincere gratitude for this feature! Debugging complex hardware is much easier now. 😊