sanny32/OpenModSim #19: Нет ответа от устройства при значения "Бит скорости 7" или " Стоповый бит 2"

artyras, 2024-09-26:
Если при условиях настройки, к примеру, 9600:8:N:1 у OmodSim и аналогичных OmodScan, я получаю сгенерированные данные. То при настройках 9600:7:N:1 или 9600:8:N:2 или 9600:7:N:2 у OmodSim,  OmodScan оповещает "Нет ответа от устройства"
[image]
консоль показывает такое

sanny32, 2024-09-26:
В данном случае консоль показывает, что приходят некорректные пакеты. Почему так происходит, надо разбираться. Пакеты отбрасывает modbus стек qt.

artyras, 2024-09-26:
Попробовал связку программ Qslave mobus emulator и  modbusPoll с настройками 9600:7:N:1 там так же на CRC ругается. Я так понимаю что для данной комбинации настроек подойдёт ASCII а не RTU, т.к. там используется LRC8  в структуре кадра сообщения вместо CRC16, но это не точно)