sanny32/OpenModSim #25: Translate

zhulinghuanyu, 2025-01-22:
pl add Translate Chinese

sanny32, 2025-02-27:
I have no Chinese language expertise. You can create pull request with omodism_cn.ts file which will be added to openmodsim project.

zhulinghuanyu, 2025-07-25:
I hope this message finds you well. This updated version includes Simplified Chinese characters and localized terms commonly used 
Please let me know if there’s anything else I can assist with or if you’d like me to make further adjustments. I’d be happy to collaborate further!
Best regards,

[omodsim_zh.zip](https://github.com/user-attachments/files/21425291/omodsim_zh.zip)

closed 2025-07-30 by sanny32

zhulinghuanyu, 2025-07-31:
Updated the translation of Simplified Chinese, fixed some translation errors, hopefully added in the next version.

[omodsim_zh_20250731.zip](https://github.com/user-attachments/files/21520477/omodsim_zh_20250731.zip)