sanny32/OpenModSim #28: Address value not saving correctly

PedroCobucci, 2025-03-14:
When you create a new project set some values and than save and open it later. Boom the values were gone

PedroCobucci, 2025-03-14:
I can open a pull request if you give me more details on how the save operation works.

sanny32, 2025-03-14:
Hello @PedroCobucci.
See [formmodsim.h](https://github.com/sanny32/OpenModSim/blob/dev/omodsim/formmodsim.h) 
`inline QDataStream& operator <<(QDataStream& out, FormModSim* frm)` and `inline QDataStream& operator >>(QDataStream& in, FormModSim* frm)` methods.

PedroCobucci, 2025-06-11:
@EvEggelen in the dev branch you wont find the omodsim.pro, at least i didn't find it. But i made a fork with the solution that also includes the old omodsim.pro file.

sanny32, 2025-06-12:
Now build system is cmake. Next release (1.8.0) with address values saving planned at next week.

closed 2025-06-16 by sanny32