sanny32/OpenModSim #33: Cannot mix incompatible Qt library (5.15.2) with this library (5.15.3)

EvEggelen, 2025-06-21:
I tried the new release ( sudo dpkg -i qt5-omodsim_1.8.0-2_amd64.deb ) on Ubuntu 22.04.5 LTS.

As some packages where missing I added the following:
sudo apt-get install libqt5serialbus5
sudo apt-get install libqt5qml5
sudo apt-get install libqt5help5

When I tried to start the application I get the following error:
:/opt/OpenModSim$ ./omodsim.sh 
Cannot mix incompatible Qt library (5.15.2) with this library (5.15.3)
Aborted (core dumped)

To my knowledge I installed 5.15.3 from the standard Ubuntu repository ( with apt-get ). Do you have a tip for me how to fix this ?

:/opt/OpenModSim$ dpkg -l | grep libqt5
ii  libqt5core5a:amd64                                          5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 core module
ii  libqt5dbus5:amd64                                           5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 D-Bus module
ii  libqt5gui5:amd64                                            5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 GUI module
ii  libqt5help5:amd64                                           5.15.3-1                                    amd64        Qt 5 help module
ii  libqt5network5:amd64                                        5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 network module
ii  libqt5printsupport5:amd64                                   5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 print support module
ii  libqt5qml5:amd64                                            5.15.3+dfsg-1                               amd64        Qt 5 QML module
ii  libqt5serialbus5:amd64                                      5.15.3-1                                    amd64        Qt serialbus - serial bus access shared library
ii  libqt5serialport5:amd64                                     5.15.3-1                                    amd64        Qt 5 serial port support
ii  libqt5sql5:amd64                                            5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 SQL module
ii  libqt5sql5-sqlite:amd64                                     5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 SQLite 3 database driver
ii  libqt5svg5:amd64                                            5.15.3-1                                    amd64        Qt 5 SVG module
ii  libqt5widgets5:amd64                                        5.15.3+dfsg-2ubuntu0.2                      amd64        Qt 5 widgets module

EvEggelen, 2025-06-21:
I also tried the following for Qt 6.6

```
pip install aqtinstall
aqt install-qt linux desktop 6.6.0 gcc_64
aqt install-qt linux desktop 6.6.0 gcc_64 -m  qtserialport
aqt install-qt linux desktop 6.6.0 gcc_64 -m qt5compat
```

In this case the library are in installed in : /opt/OpenModSim/lib
```
/opt/OpenModSim/lib$ ls
libicudata.so.56        libQt6EglFSDeviceIntegration.so.6  libQt6PrintSupport.so.6  libQt6WaylandClient.so.6
libicui18n.so.56        libQt6Gui.so.6                     libQt6Qml.so.6           libQt6WaylandEglClientHwIntegration.so.6
libicuuc.so.56          libQt6Help.so.6                    libQt6SerialBus.so.6     libQt6Widgets.so.6
libQt6Core5Compat.so.6  libQt6Network.so.6                 libQt6SerialPort.so.6    libQt6WlShellIntegration.so.6
libQt6Core.so.6         libQt6OpenGL.so.6                  libQt6Sql.so.6           libQt6XcbQpa.so.6
libQt6DBus.so.6         libQt6Pdf.so.6                     libQt6Svg.so.6
```
I also tried installing them in:
```
/opt/OpenModSim/bin/lib$ ls
libicudata.so.56        libQt6EglFSDeviceIntegration.so.6  libQt6PrintSupport.so.6  libQt6WaylandClient.so.6
libicui18n.so.56        libQt6Gui.so.6                     libQt6Qml.so.6           libQt6WaylandEglClientHwIntegration.so.6
libicuuc.so.56          libQt6Help.so.6                    libQt6SerialBus.so.6     libQt6Widgets.so.6
libQt6Core5Compat.so.6  libQt6Network.so.6                 libQt6SerialPort.so.6    libQt6WlShellIntegration.so.6
libQt6Core.so.6         libQt6OpenGL.so.6                  libQt6Sql.so.6           libQt6XcbQpa.so.6
libQt6DBus.so.6         libQt6Pdf.so.6                     libQt6Svg.so.6
```
But still no luck : /opt/OpenModSim/bin/omodsim: error while loading shared libraries: libQt6PrintSupport.so.6: cannot open shared object file: No such file or directory

EvEggelen, 2025-06-22:
Install libs and move them to the lib dir
```
aqt install-qt linux desktop 6.6.0 gcc_64 --outputdir /opt/OpenModSim/qt
mv qt/6.6.0/gcc_64/lib lib
```
This is the directory
```
:/opt/OpenModSim/lib$ ls
cmake                                  libQt6LabsSettings.prl                 libQt6QuickDialogs2.so.6.6.0
libicudata.so                          libQt6LabsSettings.so                  libQt6QuickDialogs2Utils.prl
libicudata.so.56                       libQt6LabsSettings.so.6                libQt6QuickDialogs2Utils.so
libicudata.so.56.1                     libQt6LabsSettings.so.6.6.0            libQt6QuickDialogs2Utils.so.6
libicui18n.so                          libQt6LabsSharedImage.prl              libQt6QuickDialogs2Utils.so.6.6.0
libicui18n.so.56                       libQt6LabsSharedImage.so               libQt6QuickEffects.prl
libicui18n.so.56.1                     libQt6LabsSharedImage.so.6             libQt6QuickEffects.so
libicuio.so                            libQt6LabsSharedImage.so.6.6.0         libQt6QuickEffects.so.6
libicuio.so.56                         libQt6LabsWavefrontMesh.prl            libQt6QuickEffects.so.6.6.0
libicuio.so.56.1                       libQt6LabsWavefrontMesh.so             libQt6QuickLayouts.prl
libicule.so                            libQt6LabsWavefrontMesh.so.6           libQt6QuickLayouts.so
libicule.so.56                         libQt6LabsWavefrontMesh.so.6.6.0       libQt6QuickLayouts.so.6
libicule.so.56.1                       libQt6Network.prl                      libQt6QuickLayouts.so.6.6.0
libiculx.so                            libQt6Network.so                       libQt6QuickParticles.prl
libiculx.so.56                         libQt6Network.so.6                     libQt6QuickParticles.so
libiculx.so.56.1                       libQt6Network.so.6.6.0                 libQt6QuickParticles.so.6
libicutest.so                          libQt6OpenGL.prl                       libQt6QuickParticles.so.6.6.0
libicutest.so.56                       libQt6OpenGL.so                        libQt6Quick.prl
libicutest.so.56.1                     libQt6OpenGL.so.6                      libQt6QuickShapes.prl
libicutu.so                            libQt6OpenGL.so.6.6.0                  libQt6QuickShapes.so
libicutu.so.56                         libQt6OpenGLWidgets.prl                libQt6QuickShapes.so.6
libicutu.so.56.1                       libQt6OpenGLWidgets.so                 libQt6QuickShapes.so.6.6.0
libicuuc.so                            libQt6OpenGLWidgets.so.6               libQt6Quick.so
libicuuc.so.56                         libQt6OpenGLWidgets.so.6.6.0           libQt6Quick.so.6
libicuuc.so.56.1                       libQt6PacketProtocol.a                 libQt6Quick.so.6.6.0
libQt6BundledLibjpeg.a                 libQt6PacketProtocol.prl               libQt6QuickTemplates2.prl
libQt6BundledLibpng.a                  libQt6PrintSupport.prl                 libQt6QuickTemplates2.so
libQt6Concurrent.prl                   libQt6PrintSupport.so                  libQt6QuickTemplates2.so.6
libQt6Concurrent.so                    libQt6PrintSupport.so.6                libQt6QuickTemplates2.so.6.6.0
libQt6Concurrent.so.6                  libQt6PrintSupport.so.6.6.0            libQt6QuickTest.prl
libQt6Concurrent.so.6.6.0              libQt6QmlCompiler.prl                  libQt6QuickTest.so
libQt6Core.prl                         libQt6QmlCompiler.so                   libQt6QuickTest.so.6
libQt6Core.so                          libQt6QmlCompiler.so.6                 libQt6QuickTest.so.6.6.0
libQt6Core.so.6                        libQt6QmlCompiler.so.6.6.0             libQt6QuickTestUtils.a
libQt6Core.so.6.6.0                    libQt6QmlCore.prl                      libQt6QuickTestUtils.prl
libQt6DBus.prl                         libQt6QmlCore.so                       libQt6QuickWidgets.prl
libQt6DBus.so                          libQt6QmlCore.so.6                     libQt6QuickWidgets.so
libQt6DBus.so.6                        libQt6QmlCore.so.6.6.0                 libQt6QuickWidgets.so.6
libQt6DBus.so.6.6.0                    libQt6QmlDebug.a                       libQt6QuickWidgets.so.6.6.0
libQt6DesignerComponents.prl           libQt6QmlDebug.prl                     libQt6Sql.prl
libQt6DesignerComponents.so            libQt6QmlDom.a                         libQt6Sql.so
libQt6DesignerComponents.so.6          libQt6QmlDom.prl                       libQt6Sql.so.6
libQt6DesignerComponents.so.6.6.0      libQt6QmlLocalStorage.prl              libQt6Sql.so.6.6.0
libQt6Designer.prl                     libQt6QmlLocalStorage.so               libQt6Svg.prl
libQt6Designer.so                      libQt6QmlLocalStorage.so.6             libQt6Svg.so
libQt6Designer.so.6                    libQt6QmlLocalStorage.so.6.6.0         libQt6Svg.so.6
libQt6Designer.so.6.6.0                libQt6QmlLS.a                          libQt6Svg.so.6.6.0
libQt6DeviceDiscoverySupport.a         libQt6QmlLS.prl                        libQt6SvgWidgets.prl
libQt6DeviceDiscoverySupport.prl       libQt6QmlModels.prl                    libQt6SvgWidgets.so
libQt6EglFSDeviceIntegration.prl       libQt6QmlModels.so                     libQt6SvgWidgets.so.6
libQt6EglFSDeviceIntegration.so        libQt6QmlModels.so.6                   libQt6SvgWidgets.so.6.6.0
libQt6EglFSDeviceIntegration.so.6      libQt6QmlModels.so.6.6.0               libQt6Test.prl
[... cut ...]
./omodsim: error while loading shared libraries: libQt6PrintSupport.so.6: cannot open shared object file: No such file or directory
```

```
:/opt/OpenModSim$ ./omodsim.sh 
/opt/OpenModSim/bin/omodsim: error while loading shared libraries: libQt6PrintSupport.so.6: cannot open shared object file: No such file or directory
```

Question:
Is the the file used to find the library when start OpenModSim ?
```
/opt/OpenModSim/bin$ cat qt.conf 
[Paths]
Prefix= ./../
Libraries= ./lib/
Plugins= ./plugins/
Imports= ./qml/
Translations= ./translations/
Qml2Imports= ./qml/
```

EvEggelen, 2025-06-22:
Sanny32, thanks for your support. Really appreciated.

However I deleted everything and re-installed. I did not change any configuration files or exported LD_LIBRARY_PATH. But still no luck. I does not find the libraries for to me unknown reasons. 

`sudo dpkg -i qt6-omodsim_1.8.0-2_amd64.deb`

```
/opt/OpenModSim/lib$ ls
libicudata.so.56        libQt6EglFSDeviceIntegration.so.6  libQt6PrintSupport.so.6  libQt6WaylandClient.so.6
libicui18n.so.56        libQt6Gui.so.6                     libQt6Qml.so.6           libQt6WaylandEglClientHwIntegration.so.6
libicuuc.so.56          libQt6Help.so.6                    libQt6SerialBus.so.6     libQt6Widgets.so.6
libQt6Core5Compat.so.6  libQt6Network.so.6                 libQt6SerialPort.so.6    libQt6WlShellIntegration.so.6
libQt6Core.so.6         libQt6OpenGL.so.6                  libQt6Sql.so.6           libQt6XcbQpa.so.6
libQt6DBus.so.6         libQt6Pdf.so.6                     libQt6Svg.so.6
```
```
/opt/OpenModSim$  ldd /opt/OpenModSim/bin/omodsim
	linux-vdso.so.1 (0x00007ffd64ca9000)
	libQt6PrintSupport.so.6 => not found
	libQt6SerialBus.so.6 => not found
	libQt6SerialPort.so.6 => not found
	libQt6Qml.so.6 => not found
	libQt6Network.so.6 => not found
	libQt6Help.so.6 => not found
	libQt6Widgets.so.6 => not found
	libQt6Gui.so.6 => not found
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007ef27f5cb000)
	libOpenGL.so.0 => /lib/x86_64-linux-gnu/libOpenGL.so.0 (0x00007ef27f59f000)
	libQt6Sql.so.6 => not found
	libQt6Core5Compat.so.6 => not found
	libQt6Core.so.6 => not found
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ef27f200000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ef27f4b6000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ef27f496000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ef27f491000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ef27ee00000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007ef27f148000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007ef27ecc0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ef27f618000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007ef27f465000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007ef27f45f000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007ef27f455000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007ef27f43d000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007ef27f430000)
```

```
/opt/OpenModSim$ ./omodsim.sh 
/opt/OpenModSim/bin/omodsim: error while loading shared libraries: libQt6PrintSupport.so.6: cannot open shared object file: No such file or directory
```

```
/opt/OpenModSim$ cat omodsim.sh 
#!/bin/sh

# This is default bat run script of The CQtDeployer project.
# This file contains key word that will replaced after deploy project.
#
# ####################################################################
#
# All variables has the CQT_ prefix
# BIN_PATH - are releative path to executable files of a deployed distribution.
# LIB_PATH - are releative path to libraryes of a deployed distribution.
# QML_PATH - are releative path to qml libraryes of a deployed distribution.
# PLUGIN_PATH - are releative path to qt plugins of a deployed distribution.
# BIN_PATH - are releative path to targets of a deployed distribution.

# SYSTEM_LIB_PATH - are releative path to system libraryes of a deployed distribution.
# BASE_NAME - are base name of the executable that will be launched after run this script.
# CUSTOM_SCRIPT_BLOCK - This is code from the customScript option
# RUN_COMMAND - This is command for run application. Requred BASE_DIR variable.
#
# ####################################################################

BASE_DIR=$(dirname "$(readlink -f "$0")")
export PATH="$BASE_DIR"/bin/:$PATH
export LD_LIBRARY_PATH="$BASE_DIR"/lib/:"$BASE_DIR":$LD_LIBRARY_PATH
export QML_IMPORT_PATH="$BASE_DIR"/qml/:$QML_IMPORT_PATH
export QML2_IMPORT_PATH="$BASE_DIR"/qml/:$QML2_IMPORT_PATH
export QT_PLUGIN_PATH="$BASE_DIR"/plugins/:$QT_PLUGIN_PATH
export QTWEBENGINEPROCESS_PATH="$BASE_DIR"/bin//QtWebEngineProcess
export QT_QPA_PLATFORM=xcb
export QTDIR="$BASE_DIR"
export CQT_PKG_ROOT="$BASE_DIR"
export CQT_RUN_FILE="$BASE_DIR/omodsim.sh"

export QT_QPA_PLATFORM_PLUGIN_PATH="$BASE_DIR"/plugins//platforms:$QT_QPA_PLATFORM_PLUGIN_PATH

"$BASE_DIR/bin/omodsim" "$@" 
```

sanny32, 2025-06-22:
The problem at deb package is approved and will be fixed.
As a temporary solving you must execute `sudo setcap 'cap_net_bind_service=-ep' /opt/OpenModSim/bin/omodsim`
Then you can execute `omodsim.sh`. 
If error with xcb will be occurred, please install `sudo apt install libxcb-cursor0`

sanny32, 2025-06-23:
Fixed with [qt5-omodsim_1.8.0-3_amd64.deb ](https://github.com/sanny32/OpenModSim/releases/download/1.8.0/qt5-omodsim_1.8.0-3_amd64.deb) and [qt6-omodsim_1.8.0-3_amd64.deb](https://github.com/sanny32/OpenModSim/releases/download/1.8.0/qt6-omodsim_1.8.0-3_amd64.deb) packages.

EvEggelen, 2025-06-25:
I tried the qt6-omodsim_1.8.0-3_amd64.deb and got the following error:
```
:/opt/OpenModSim$ ./omodsim.sh 
qt.qpa.plugin: From 6.5.0, xcb-cursor0 or libxcb-cursor0 is needed to load the Qt xcb platform plugin.
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/opt/OpenModSim/plugins//platforms:" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, offscreen, vkkhrdisplay, eglfs, wayland, minimalegl, vnc, wayland-egl, minimal, linuxfb.
```

I can confirm that the qt5 deb packages work. Thanks !

sanny32, 2025-06-25:
See upper `sudo apt install libxcb-cursor0`

EvEggelen, 2025-06-25:
Thanks for the tip. I missed it. Is it possible to add this as a dependency in the deb files ? Or would that cause other issues ?

sanny32, 2025-06-25:
Please create other issue for package dependencies.