sanny32/OpenModSim #37: Add option to respond with zero values for unemulated registers

LepreconSTR, 2025-07-21:
**Current behavior**
Currently, when a request is made to a register that is not in the list of emulated registers, the program returns an “ILLEGAL CODE” error.

**Problem description**
This behavior creates unnecessary complexity during device emulation, as it forces users to emulate all possible registers even if only a subset is required for testing purposes.

**Proposed solution**
Introduce a configurable option that allows the system to respond with zero values (register = 0) for unemulated registers instead of returning an “ILLEGAL CODE” error.

sanny32, 2025-07-21:
This can be implemented.

sanny32, 2025-07-22:
Will be configured via Display Definition dialog
[image]

commit a5343ee1 (2025-07-22): fixed #37
closed 2025-07-30 by sanny32